dchristl/macless-haystack #107: Can not save any accessory when user new built a release apk.

cuonghm90, 2024-06-17:
As you know, I'm developing new feature about reading battery level. (https://github.com/dchristl/macless-haystack/issues/100)
So when I build a new apk, in **release mode**, I can not save any accessory.
Step to reproduce
1. Build app: "flutter build apk" or "flutter run --release"
2. Add new accesory.
3. Press "Back" of device to exit to home screen.
4. Open App
-> lost devices added.

Can you check that because I don't much about flutter.
Many thanks.

dchristl, 2024-06-22:
You have to press "import" (in the right top corner) instead of back to import new accessories

cuonghm90, 2024-06-24:
I imported accessory in step "2. Add new accesory."

The extract step to procedure is
1. Build app: "flutter build apk" or "flutter run --release"
2. Add new accessory/Import accessory
3. Press "Back" button of android devices to exit to home screen.
4. Open App again
-> lost devices added.

dchristl, 2024-07-29:
Sorry, but how am I supposed to help you here? You should already have a basic understanding of flutter and/or Android development. I would start with adb (logcat) or debugging the app. Flutter is quite chatty when an error occurs.