dchristl/macless-haystack #113: Failing To Receive 2FA Request

CelicaXX, 2024-08-01:
Not entirely sure what's going on. I'm not receiving any errors in the terminal after running:
```
docker run -it --restart unless-stopped --name macless-haystack -p 6176:6176 --volume mh_data:/app/endpoint/data --network mh-network christld/macless-haystack
``` 

I have verified that 2FA is indeed enabled and am receiving 2FA codes when logging into the iCloud website through a browser as well as all other methods. I've waited for appx. an hour and not received a code.

dchristl, 2024-08-05:
This is sadly a well known problem and is already discussed in #46 . There is nothing I can do about it, only Apple know why some accounts are working and some not. You can try some workarounds, discussed in the liniked issue.

zerog2k, 2024-09-18:
for me at least #125  is fixing this problem
(I have multiple 2FA numbers registered - the first one was phone [not sms], and only second number has sms)

Styne13, 2024-11-13:
Hi,
thanks for all your work integrating this project into a docker environment :)

I have also an issue with 2FA.

@zerog2k: Can you please describe the modification you made to #125 or even update the pr?

I also tried out [Find my Flipper](https://github.com/MatthewKuKanich/FindMyFlipper) in a non-Docker environment and for the pypush_gsa_icloud.py in there to login works.

Thanks and Best Regards

Marcel

CoderXueXue, 2024-12-29:
我也遇到了，我通过重新注册了一个账号，然后将账号内只留有一个信任的电话号码，没有信任的设备 就可以收到了。

jannisko, 2025-01-15:
For me #125 made it work like a charm. @Styne13 to patch it in I just exec'd into the container `docker exec -it macless-haystack /bin/bash` and edited the file `nano endpoint/register/pypush_gsa_icloud.py`, replacing line 241 with the new lines [236-242 from the pr](https://github.com/dchristl/macless-haystack/pull/125/files#diff-02eae05a7319b9736412a6d4189fc5e822a955b17610e54fc1cd2aaff8dfd749R236-R242).
I then ran `python endpoint/mh_endpoint.py` in the container to run through the authentication. After it is done, you can exit the container and restart it so it runs in the background using the saved auth info.

dchristl, 2025-02-06:
The fix in #125 will be included in the next release.

closed 2025-02-06 by dchristl