dchristl/macless-haystack #114: iOS App

isibizi, 2024-08-05:
Hello,

I have found a nice iOS app for this project. The only problem is that the developer from the other project uses their own version of the macless-haystack server. The server from the other project is a little bit buggy. I have already tested it, but when I import a key, the tags are only trackable one time. After that, no new coordinates are shown. I don't have experience with Swift, but the developer mentioned that it is simple to change the server to request the geo-coordinates from macless-haystack. If someone has experience adapting the iOS app, we would have a great UI for tags

https://github.com/rkreutz/openhaystack-ios

https://github.com/rkreutz/openhaystack-server

T-REX-XP, 2024-09-04:
Do you really need the iOS app? The AirTags has native support on Apple devices. If it so expensive, you can by China version of Air Tags that are fully compatible with Apple Find my app.  They costs starting from 3-5 $.
For instance:
https://es.aliexpress.com/w/wholesale-air-tag-find-my.html?spm=a2g0o.productlist.search.0

isibizi, 2024-09-04:
Sure, but when I use a tag, like an AirTag, to track a car or a bike, the thief could see that there’s an AirTag nearby.

dchristl, 2024-09-06:
Hello,

the project is aimed at users who don't have a Mac (like me) and still want to use Openhaystack. There are a few things that work much easier and more natively with a Mac and are bypassed here (e.g. the Anisette server is not needed at all). 
Nevertheless, the frontend is written in Flutter and would actually only need to be compiled for the Mac. Flutter supports this out of the box. I just can't try it out because I don't have a Mac. 

Kind Regards,
Danny