dchristl/macless-haystack #116: It is Possible to emuate ADV key on Windows Laptops

kiam001, 2024-08-21:
Maybe some Docs from micrsoft helps for this issue:

https://learn.microsoft.com/en-us/windows/uwp/devices-sensors/ble-beacon

dchristl, 2024-09-06:
Hello,

sorry, I don't understand what you want to say and achieve with this. 

Regards
Danny

kiam001, 2024-09-06:
On Windows, create a public key to find your laptop via the Bluetooth le module

kiam001, 2024-09-06:
Why shouldn't that be part of the project?

I mean you can create keys with the Key Generator Python, and there is already a script for Linux "HCI.py" it would be cool to be able to find windows devices via macless hayastack over the Bluetooth interface that are in a Windows laptop.

dchristl, 2024-09-09:
> Why shouldn't that be part of the project?

Where did I say that? I just can't make any reference to this project (or OPenHaystack, Apple) based on your confused and inane statements, and you're not answering my questions properly either. 

The HCI.py script in Linux is merely a proof of concept and has no practical significance. The PC must be permanently switched on for the script to run. That's why I'm generally wondering why you should use it (except perhaps on a Raspberry). Is there already an Openhaystack-compatible project for the interface you linked (Bluetooth LE) or a Windows port of the HCI.py script (that's pyton and should be easy to do)? If so, this would already be possible for tracking with macless-haystack. I can't find a project or a proof of concept for this.

lovelyelfpop, 2025-03-26:
A better solution is to connect a Bluetooth chip with haystack firmware in parallel to the positive and negative poles of the laptop battery.