dchristl/macless-haystack #118: Help for support asr5601a

kiam001, 2024-08-24:
Hello, its possible to flash this on this soc https://www.asrmicro.com/en/goods/download?cid2=21

[
[image]
[image]
[image]

pix, 2024-08-26:
I have the exact same tag, I tried contacting ASR for a SDK with no luck.

dchristl, 2024-09-06:
Unfortunately, I can't find any firmware on the net that can be patched here. I would therefore say that it is currently not possible

pix, 2024-09-11:
The SDK isn't freely available. Plenty of docs, but no binaries / SDK  / example code.