dchristl/macless-haystack #119: Home Assitant / NodeRED integration?

zjonesz, 2024-08-25:
Is this possible to do? Is there an API?

dchristl, 2024-09-06:
Hello,
no, there is no api for that and other projects are better suited for that. For example [biemster's FindMy](https://github.com/biemster/FindMy)

Regards,
Danny

iMartyn, 2025-01-24:
To expand this answer a little, because it can be tricky to understand the moving parts :

- macless-haystack is an API for the webapp/mobile app but this isn't an appropriate place to integrate with home-assistant or other system.
- macless-haystack talks to anisette which is an API that talks to apple's servers.  Anisette is an approximately appropriate place to integrate with home-assistant and good news, [someone already has](https://github.com/malmeloo/hass-FindMy).

If you have macless-haystack running, you already have the parts needed of FindMy mentioned above, so you simply need to install that integration.  Using both is not a problem, and the hass integration is only checking for changes infrequently.

closed 2025-02-05 by dchristl