dchristl/macless-haystack #120: Alternative firmwares for nrf52810 and nrf51822

pix, 2024-08-26:
Hi, shameless plug here: https://github.com/pix/heystack-nrf5x

I managed to convert an old Tile tag (with an nrf52810) to Haystack.
I used alcatrava's firmware as a base too and fixed some potential memory leaks and updated the SDK / Makefiles.

The code is running since two days, without problems for now.

dchristl, 2024-09-06:
Hello,

Thanks for the info, if I ever have time I can include this as an alternative in the documentation. Does your firmware support changing keys/mac?

Kind Regards,
Danny

pix, 2024-09-11:
Yes. More than 4500 keys on a nrf51 :-)

You might want to wait until this [issue](https://github.com/pix/heystack-nrf5x/issues/3) is ironed out. I have a branch for a fix, but I'm waiting for new hardware to test things out.

dchristl, 2024-09-14:
Sounds great. Did you happen to take measurements of the power consumption? I had the problem with the firmware that the power consumption was significantly higher with changing keys. By significant I mean about 10-15 times shorter lifetime compared to 1 key to several. This was probably due to the wake up (key - Mac change) and I couldn't solve this with the Nordic forum either. That's why I no longer use my firmware, because changing the battery every month is annoying and expensive

pix, 2024-09-17:
Nope, not at the moment. I'm looking for opportunities to get my hands on a oscilloscope :)

ctandi, 2024-09-20:
> Did you happen to take measurements of the power consumption?

I did, with the Nordic Power Profiler Kit 2 :)

The HolyIoT YJ-16013 (nRF52832) sitting on my table right now is running with an average current draw of 14 µA (2000 ms Advertising Interval, 4 dBm, 80 Keys, 900 sec. rotation interval). I tried setting it to the external 32 KHz oscillator, which I heard could reduce the power consumption even further. However when I edit the sdk_config.h the SoC crashes with

```
[nrf52.cpu] halted due to breakpoint, current mode: Thread 
xPSR: 0x81000000 pc: 0x00027068 msp: 0x2000ffc0
```

The HolyIoT 17095 (nRF52832) averages at around 20 µA with the same settings as the 16013. It doesn't have an XTAL.

A HolyIoT 21014 (nRF52810) with 60 keys (rotating every 900 secs) advertising every 3000 ms takes around 10 µA. I also haven't tested using the XTAL instead of the internal source with this model yet.

The no-name beacon with an nRF51822 that I have draws 18 µA with the internal RC circuit and 16 µA with the XTAL in use.

pix, 2024-09-20:
Thanks for the measurments. I got my hand on a OWON-VDS1022.

I got my first snaps of the sending and key rotation power consumption:
[image]

It's been years since I touched an oscilloscope, and my shunt resistor is, well from the back of the drawer, but i'll be able to validate my measurement skills with your input ! Thanks :)

pix, 2024-09-20:
> A HolyIoT 21014 (nRF52810) with 60 keys (rotating every 900 secs) advertising every 3000 ms takes around 10 µA. I also haven't tested using the XTAL instead of the internal source with this model yet.

If you have optimized settings XTAL settings for the HolyIoT (or other), I could add the relevant board headers and flags on the project to build them more easily. That would be great :+1:

vasimv, 2024-10-16:
NRF51/52 has switching DC-DC regulator (to efficiently convert input voltage into lower voltage the microcontroller uses), it does reduce power consumption greatly if input voltage is more than 2 volts (in comparison with default LDO regulator). It saves much more power than installing 32.768kHz quartz, but requires inductor and few capacitors on the board. Most of boards have the inductor, so you have just enable the DCDC regulator to lower power consumption significally.

pix, 2024-10-16:
Do you have any link / example code / sdk config showing how to enable it ? If it's available on both I can build both versions (unless for boards don't supporting it)

pix, 2024-10-16:
FYI: I added dcdc build variants (I didn't enable it for yj17024 as low power consumption may not be the target for this board, I might be proven wrong)

`make all` at the root will build all the variants.

vasimv, 2024-10-16:
Ugh, not sure how you set it in old nrf52 sdk, sorry. For zephyr SDK this is "CONFIG_BOARD_ENABLE_DCDC=y" in prj.conf at compile time. Probably, "#define POWER_CONFIG_DEFAULT_DCDCEN 1" for enabling by default.