dchristl/macless-haystack #122: Fixed mac address

Koxx3, 2024-09-08:
Hello,
thanks for this awesome project !
I need to create a custom version which allow me to use a custom fixed mac address.
If I understand correctly, I need the public key to match this address ?
Any idea how I could do that ? how to generate keys with a given mac ?
Thanks !

dchristl, 2024-09-09:
Hello,

can you perhaps explain why you need this? You don't necessarily need the public key, this is only the case with the original AirTag. Here the key rotates and parts of the Mac are determined from the key to prevent movement profiles from being recorded. With the OpenHaystack solutions, the Mac is not set at all (the original Mac is used) or is static. 

Kind Regards,
Danny

Koxx3, 2024-09-09:
Thanks for your answer.

I have ESP32 devices which already have fixed generated mac address.
I need to keep those addresses all the time to allow phones to connect to those devices.
So, what I'd like is to integrate the haystack/openhaystack solution to make those ESP32 devices recognized as "tags" and allow me to track them with your web portal (or custom app).

I am not sure I understand your answer. The original openhaystack seems to already have a matching public key with mac address and advertisement payload (esp_ble_gap_set_rand_addr) : https://github.com/seemoo-lab/openhaystack/blob/main/Firmware/ESP32/main/openhaystack_main.c

If I don't force the mac address with set_addr_from_key, it doesn't seems to work.
If I do, it works perfectly.

Did I miss something ?

dchristl, 2024-09-14:
> I am not sure I understand your answer. The original openhaystack seems to already have a matching public key with mac address and advertisement payload (esp_ble_gap_set_rand_addr / esp_ble_gap_set_rand_addr) :
Did I miss something ?

Sorry, I mixed up 2 projects in my head without checking. The Mac/Bluetooth address is of course set everywhere, in the original firmware it only rotates additionally. 
Unfortunately, I can no longer find the openHaystack documentation that describes which part changes. However, since the Mac is (almost) completely determined from the public key, your requirements will not be met. At least not with a fixed address. It would only be possible to generate a key and store the resulting Mac on all phones.

Perhaps you can explain why the connection is only necessary with a fixed Mac. There may be other options here. Alternating Macs are quite common with some devices to create privacy.