dchristl/macless-haystack #123: When the number of location data reaches 21, it is no longer updated

Echoooo0827, 2024-09-12:
The number of positioning data is no longer updated after 21, the device is not shut down, but after reaching 21, it can only obtain 21 after requesting refresh all the time.
Please help to answer, thank you very much!

Echoooo0827, 2024-09-12:
A maximum of 21 location data can be obtained

dchristl, 2024-09-14:
Perhaps there are only 21 reports on the Apple server.  Sometimes these are time-delayed. If you go to an area with a lot of people, new reports should appear. It is not necessary to reload the firmware. There are no limitations in the firmware or when retrieving the reports.

Echoooo0827, 2024-09-14:
Okay, thank you so much!

Echoooo0827, 2024-09-14:
> Perhaps there are only 21 reports on the Apple server. Sometimes these are time-delayed. If you go to an area with a lot of people, new reports should appear. It is not necessary to reload the firmware. There are no limitations in the firmware or when retrieving the reports.也许 Apple 服务器上只有 21 个报告。有时这些是时间延迟的。如果您去的人很多的区域，应该会出现新的报告。无需重新加载固件。固件或检索报告时没有限制。

> Perhaps there are only 21 reports on the Apple server. Sometimes these are time-delayed. If you go to an area with a lot of people, new reports should appear. It is not necessary to reload the firmware. There are no limitations in the firmware or when retrieving the reports.也许 Apple 服务器上只有 21 个报告。有时这些是时间延迟的。如果您去的人很多的区域，应该会出现新的报告。无需重新加载固件。固件或检索报告时没有限制。

However, I have put it in the place where I can send the location, and there are Apple phones around, but it will not be updated. I re-brush the firmware and test again, but it is still 21