dchristl/macless-haystack #124: Debugging beacon not sending location

M-Gregoire, 2024-09-15:
Hello 👋 ,

First of all, thank you for spending time on this project, this looks awesome.

I'm trying to use it with a nRF51822 (bought of Aliexpress). I've generated the keys, injected them into the firmware (28 bytes moved with `dd`), solder the pins (Following pinout from https://github.com/seemoo-lab/openhaystack/issues/35#issuecomment-828338751) and used `openocd` to flash it:

```
$ openocd -f openocd.cfg -c "init; halt; nrf51 mass_erase; program nrf51_firmware_prefix.bin; reset; exit"
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.166882
Info : [nrf51.cpu] Cortex-M0 r0p0 processor detected
Info : [nrf51.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for nrf51.cpu on 3333
Info : Listening on port 3333 for gdb connections
[nrf51.cpu] halted due to debug-request, current mode: Handler HardFault
xPSR: 0xc0000003 pc: 0xfffffffe msp: 0xffffffe0
Info : nRF51822-QFAA(build code: H0) 256kB Flash, 16kB RAM
Info : Mass erase completed.
Info : A reset or power cycle is required if the flash was protected before.
[nrf51.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Warn : Adding extra erase range, 0x00024d78 .. 0x00024fff
** Programming Finished **
```

So overall seems like a success to me. After setting up the docker containers, going to the WebUI and importing the JSON, my device is never picked up, even after multiple hours and going out in a city with it, it pulls `0 location(s)`.

I've checked the battery and tried to power it with USB rather than the cell battery just to make sure. LED turns on only during the flashing but not after, but I'm guessing this is expected with this firmware. I did power off - power on the device after flashing.

I've tried multiple BLE scanners on my phone but I'm not even sure what the MAC address is for my beacon.

In summary, despite all the steps looking like success, I don't receive any location data, and I'm unsure how to debug this problem. I wasn't able to find a way to get logs through the ST-LINK nor other way to check whether the tag is working properly.

Any kind of help you be greatly appreciated.

M-Gregoire, 2024-09-15:
I just tried flashing an ESP32:
```
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
```

And it was detected, so at the least that confirms the rest of the infrastructure is working. I've also understood that the MAC address is set by the firmware the match the key, so no use of knowing the MAC address of the device.

I guess being able to get the MAC address from the `firmware.bin` would be useful though.

zerog2k, 2024-09-18:
depending upon whether your board has LF crystal (32k watch crystal) or not, you may need to try `BOARD=BOARD_ALIEXPRESS_NO_XTAL`

pix, 2024-09-18:
> I've tried multiple BLE scanners on my phone but I'm not even sure what the MAC address is for my beacon.

You can take a look at this to show the expected MAC for a keyfile: https://github.com/pix/heystack-nrf5x/blob/master/tools/showmac.py

I use an app called "BLE Scanner" and filter by RSSI < 50 or so.

M-Gregoire, 2024-09-25:
Thanks, this is exactly what I was looking for.

I'll try `BOARD_ALIEXPRESS_NO_XTAL` and report back, but I thought this wasn't needed as the `BOARD_SIMPLE` used by default mentions:
```
 BOARD_SIMPLE is the default board with no external crystal to maximize compatibility
```

So I'm not sure to understand the difference between the two boards.

M-Gregoire, 2024-09-26:
Update: I've compiled with `BOARD_ALIEXPRESS_NO_XTAL` and was able to make my beacon work, thanks a lot!

I initially used `showmac.py` on the binary rather than the keyfile which printed out `5501` results but quickly realized my mistake and was also able to get the MAC address.

So everything is perfect and working as expected! Thanks a lot helping me through this.

closed 2024-09-26 by M-Gregoire