dchristl/macless-haystack #126: my macless-haystack server stopped working ;(

Juzjestem, 2024-09-19:
Hi, my haystack stopped working, I did a new installation and registration with Apple but nothing helped, what could be the cause?

2024-09-19 11:29:48 172.18.0.1 - - [19/Sep/2024 09:29:48] "POST / HTTP/1.1" 200 -
2024-09-19 11:29:51 2024-09-19 09:29:51,651 - DEBUG - Getting with post: b'{"ids":["it4ZvLuzGM5WPId/nwTkI8geDlYLNc5K7+wwMHZ4wyY="],"days":1}'
2024-09-19 11:29:51 2024-09-19 09:29:51,651 - DEBUG - Querying for 1 days
2024-09-19 11:29:51 2024-09-19 09:29:51,958 - DEBUG - Return from fetch service:
2024-09-19 11:29:51 2024-09-19 09:29:51,958 - DEBUG - {"results":[],"statusCode":"200"}

Andrez092, 2024-09-19:
try this
https://mega.co.nz/#!B31G2LaZ!uBBOCp9hLC7bq9kP8NC6s4DanQJJZuoFRr5FI_jkeic
I put the necessary dlls in the archive

Juzjestem, 2024-09-20:
NOW:
2024-09-20 08:45:23 172.18.0.1 - - [20/Sep/2024 06:45:23] "POST / HTTP/1.1" 501 -
2024-09-20 08:45:23 2024-09-20 06:45:23,615 - DEBUG - Getting with post: b'{"ids":["ZMWmtbxcAxAHrJQlNhpLQR+QGWcZnKa9h9qRO2K5IWQ="],"days":2}'
2024-09-20 08:45:23 2024-09-20 06:45:23,615 - DEBUG - Querying for 2 days
2024-09-20 08:45:24 2024-09-20 06:45:24,075 - DEBUG - Return from fetch service:
2024-09-20 08:45:24 2024-09-20 06:45:24,075 - DEBUG - 
2024-09-20 08:45:24 2024-09-20 06:45:24,075 - ERROR - Unknown error occured Expecting value: line 1 column 1 (char 0)
2024-09-20 08:45:24 Traceback (most recent call last):
2024-09-20 08:45:24   File "/app/endpoint/mh_endpoint.py", line 104, in do_POST
2024-09-20 08:45:24     result = json.loads(r.content.decode())
2024-09-20 08:45:24              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-09-20 08:45:24   File "/usr/local/lib/python3.12/json/__init__.py", line 346, in loads
2024-09-20 08:45:24     return _default_decoder.decode(s)
2024-09-20 08:45:24            ^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-09-20 08:45:24   File "/usr/local/lib/python3.12/json/decoder.py", line 337, in decode
2024-09-20 08:45:24     obj, end = self.raw_decode(s, idx=_w(s, 0).end())
2024-09-20 08:45:24                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-09-20 08:45:24   File "/usr/local/lib/python3.12/json/decoder.py", line 355, in raw_decode
2024-09-20 08:45:24     raise JSONDecodeError("Expecting value", s, err.value) from None
2024-09-20 08:45:24 json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
2024-09-20 08:45:24 172.18.0.1 - - [20/Sep/2024 06:45:24] "POST / HTTP/1.1" 501 -

cielaton, 2024-09-27:
Completely remove the macless-haytack docker container:
`docker stop macless-haystack`
`docker rm macless-haystack`
`docker volume rm mh_data`
Then rerun the container, login and it will be able to query the information.

aramix, 2024-12-25:
@cielaton I have the same issue after running the container for about 30 mins, why is this happening?

cielaton, 2024-12-25:
@aramix  Just making sure the Anisette container is up and running **before** the Macless-haystack container. Whenever I reboot my server, I have to clean the containers and start them from scratch.