dchristl/macless-haystack #141: auth.json token expired?

Philip2809, 2024-11-05:
How long should the device registration live on? I registered the macless-haystack on the 30th of October and today it stopped working. Removing the auth.json and redoing the registration it now works again, any idea why and how long the auth.json is valid? The response from the fetch was empty and the json.loads was not able to parse the empty string the server sent.

dchristl, 2025-02-05:
Hello,

that never runs out. I have had the same auth.json for more than 1 year. You may have removed the device from your Apple account or not used it for too long.

supaeasy, 2025-02-08:
Every time I update anisette server I have to delete auth and re-enter Credentials/2FA. After that everything works fine but it gives the weirdest errors beforehand.

dchristl, 2025-02-09:
> Every time I update anisette server I have to delete auth and re-enter Credentials/2FA. After that everything works fine but it gives the weirdest errors beforehand.

Did you see the small edit in the readme of the docker command? ([explanation](https://github.com/Dadoum/anisette-v3-server/pull/29)) Recreating the Anisette container should solve your problem

supaeasy, 2025-02-09:
I recreated it this way and now, for the first time there are files inside the mapped location. So I guess the problem will be gone from now on. Will see with the next update if this does the trick. Seems promising.

supaeasy, 2025-02-10:
well, I just tested it: I have to delete auth.json and re-enter my credentials but 2FA ist not necessary anymore. Weird but at least I know the trick now.

supaeasy, 2025-02-10:
I seem to have it working now for good. Tested this including purging and deleting images. Login data persists. I was using a different network than standard configuration before, maybe that was a problem though that would be strange. Maybe it also has something to do with the Synology DSM I am running it on; who knows? I felt like writing this down just in case anyone else has this problem. Was thinking about a pull request to the readme but don't really know where to put this or if this is appreciated.

So for everyone using Synology DSM and Container Manager here is my docker-compose file that works fine for me. There is just 3 things to do in total:

- create a folder in a location of your choice with folders [name1] and [name2]
- create a new project in Container Manager using docker-compose.yaml below, replacing [name1] & [name2] with their filesystem location (format would be something like this example: **/volume1/docker/mh/anisette**) and start it
- Open macless-haystack project-> click macless-haystack -> in the upper right click open Terminal. A completely blank shell will open. Enter your AppleID and hit enter. Enter password and when asked to enter 2FA.

**docker-compose.yaml**
```
name: macless-haystack
services:
    anisette-v3-server:
        restart: unless-stopped
        container_name: anisette
        ports:
            - 6969:6969
        volumes:
            - [name1]:/home/Alcoholic/.config/anisette-v3
        image: dadoum/anisette-v3-server
        networks:
            - mh-network
    macless-haystack:
        stdin_open: true
        tty: true
        restart: unless-stopped
        container_name: macless-haystack
        ports:
            - 6176:6176
        volumes:
            - [name2]:/app/endpoint/data
        image: christld/macless-haystack
        networks:
           - mh-network

networks:
  mh-network:
```

Yxini-p, 2025-05-26:
> 现在看来它已经正常工作了。我测试了这个问题，包括清除和删除图片。登录数据仍然存在。我之前使用的网络与标准配置不同，这可能是个问题，不过那样会很奇怪。也许这也与我运行它的 Synology DSM 有关；谁知道呢？我想把这个问题写下来，以防其他人也遇到同样的问题。我考虑过对 readme 进行拉取请求，但不知道该放在哪里，或者不知道是否会有人喜欢。最终，这只是将 readme 中的标准配置转换为 yaml 文件，并针对此特定用例添加了“空白 shell 的知识”。
> 
> 因此，对于所有使用 Synology DSM 和 Container Manager 的用户，这里是我的 docker-compose 文件，它对我来说运行良好。总共只需做三件事：
> 
>     * 在您选择的位置创建一个文件夹，其中包含子文件夹 [name1] 和 [name2]
> 
>     * 使用下面的 docker-compose.yaml 在容器管理器中创建一个新项目，将 [name1] 和 [name2] 替换为其文件系统位置（格式类似于此示例： ** /volume1/docker/mh​​/anisette ** ）并启动它
> 
>     * 打开 macless-haystack 项目->点击 macless-haystack->在右上角点击“打开终端”。一个完全空白的 shell 将会打开。输入你的 AppleID 并按回车键。输入密码，并在提示时输入 2FA。
> 
> 
> **docker-compose.yaml **
> 
> ```
> name: macless-haystack
> services:
>     anisette-v3-server:
>         restart: unless-stopped
>         container_name: anisette
>         ports:
>             - 6969:6969
>         volumes:
>             - [name1]:/home/Alcoholic/.config/anisette-v3
>         image: dadoum/anisette-v3-server
>         networks:
>             - mh-network
>     macless-haystack:
>         stdin_open: true
>         tty: true
>         restart: unless-stopped
>         container_name: macless-haystack
>         ports:
>             - 6176:6176
>         volumes:
>             - [name2]:/app/endpoint/data
>         image: christld/macless-haystack
>         networks:
>            - mh-network
> 
> networks:
>   mh-network:
> ```

I use Docker Compose Macross Haystack. Click on Macross Haystack in the upper right corner and open the terminal. No feedback on input tasks; Can you provide more detailed instructions