dchristl/macless-haystack #142: Local volume path cannot be used

M-Gregoire, 2024-11-05:
Hello :wave: ,

I've encountered a weird bug where using a local volume instead of a docker volume prevents the `config.ini` from being created.

**Works as expected:**
```
docker run -it --restart unless-stopped --name macless-haystack -p 6176:6176 --volume mh_data:/app/endpoint/data --network mh-network christld/macless-haystack
```

Replacing `--volume mh_data:/app/endpoint/data` with `--volume /tmp/mh-data:/app/endpoint/data` leads to : 

```
2024-11-05 15:09:34,756 - INFO - No auth-token found.
2024-11-05 15:09:34,757 - INFO - Trying to register new device.
2024-11-05 15:09:34,757 - INFO - Trying to login
Apple ID: test@test.fr
Password: 
2024-11-05 15:09:40,060 - INFO - Authentication request initialization
Traceback (most recent call last):
  File "/app/endpoint/mh_endpoint.py", line 159, in <module>
    apple_cryptography.registerDevice()
  File "/app/endpoint/register/apple_cryptography.py", line 77, in registerDevice
    getAuth(regenerate=True)
  File "/app/endpoint/register/apple_cryptography.py", line 50, in getAuth
    mobileme = icloud_login_mobileme(
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 40, in icloud_login_mobileme
    g = gsa_authenticate(username, password)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 76, in gsa_authenticate
    r = gsa_authenticated_request(
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 134, in gsa_authenticated_request
    "Request": {"cpd": generate_cpd()},
                       ^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 168, in generate_cpd
    cpd.update(generate_anisette_headers())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 173, in generate_anisette_headers
    h = json.loads(requests.get(config.getAnisetteServer(), timeout=5).text)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/config.py", line 22, in getAnisetteServer
    value = config.get('Settings', 'anisette_url')
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/configparser.py", line 759, in get
    d = self._unify_values(section, vars)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/configparser.py", line 1132, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'Settings'
```

I've tried changing the permissions of the folder but the result seems to always be the same.

M-Gregoire, 2024-11-05:
Side note: I tried using `docker exec -it macless-haystack bash` to `touch /app/endpoint/data/config.ini` and it works, so I don't believe it's a permission issue.

dchristl, 2025-02-05:
Hello,

what you are doing here is mounting a volume with the name /tmp/mh-data at the address. Since this does not exist, an empty volume will probably be created. This is why touch (empty file) works but writing does not.  The folder is also not created under /tmp, right?

Read the [Docker doc ](https://docs.docker.com/engine/storage/volumes/) how to change volume locations.