dchristl/macless-haystack #145: Webapplication load error

lovelyelfpop, 2024-12-03:
The page can show when I first open the webapplication, but it stuck at the logo when I refresh the page.
Tried chrome 130, chrome 118, firefox 133, all the same issue
[image]

dchristl, 2025-02-05:
Hello,
the web application is currently running perfectly. Are you still having problems? If so, delete the browser cache and/or start with deactivated plugins.

lovelyelfpop, 2025-02-20:
Is it only support https? There is an error when I serve the web application in http?

[image]

dchristl, 2025-02-23:
What do you mean by that? The Web-frontend on GitHub is always https. If you host your own, don't do that, if you don't know what you are doing. You have to change several things inside to make it work and I also don't support own hosting. I have explained [here](https://github.com/dchristl/macless-haystack/blob/main/FAQ.md#how-can-i-host-my-own-web-frontend) why this is not necessary at all.