dchristl/macless-haystack #148: Failure in name resolution error when attempting to use apple ID

mearsy, 2024-12-12:
I have been succesful in getting this up and running on a Mac, using docker but when I get to the step of logging in on a new Linux box ive set up I get this weird name resolution failure.

Can anyone help please?

From https://github.com/dchristl/macless-haystack
 * branch            main       -> FETCH_HEAD
2024-12-12 11:23:06,218 - DEBUG - Searching for token at /app/endpoint/data/auth.json
2024-12-12 11:23:06,220 - INFO - No auth-token found.
2024-12-12 11:23:06,221 - INFO - Trying to register new device.
2024-12-12 11:23:06,222 - INFO - Trying to login
Apple ID: ameliarosemears@gmail.com
Password: 
2024-12-12 11:23:26,571 - INFO - Authentication request initialization
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 198, in _new_conn
    sock = connection.create_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/util/connection.py", line 60, in create_connection
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/socket.py", line 963, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
socket.gaierror: [Errno -3] Temporary failure in name resolution

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 793, in urlopen
    response = self._make_request(
               ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 496, in _make_request
    conn.request(
  File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 400, in request
    self.endheaders()
  File "/usr/local/lib/python3.12/http/client.py", line 1331, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/local/lib/python3.12/http/client.py", line 1091, in _send_output
    self.send(msg)
  File "/usr/local/lib/python3.12/http/client.py", line 1035, in send
    self.connect()
  File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 238, in connect
    self.sock = self._new_conn()
                ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 205, in _new_conn
    raise NameResolutionError(self.host, self, e) from e
urllib3.exceptions.NameResolutionError: <urllib3.connection.HTTPConnection object at 0x7d6be11435f0>: Failed to resolve 'anisette' ([Errno -3] Temporary failure in name resolution)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/requests/adapters.py", line 486, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 847, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/util/retry.py", line 515, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='anisette', port=6969): Max retries exceeded with url: / (Caused by NameResolutionError("<urllib3.connection.HTTPConnection object at 0x7d6be11435f0>: Failed to resolve 'anisette' ([Errno -3] Temporary failure in name resolution)"))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/endpoint/mh_endpoint.py", line 159, in <module>
    apple_cryptography.registerDevice()
  File "/app/endpoint/register/apple_cryptography.py", line 77, in registerDevice
    getAuth(regenerate=True)
  File "/app/endpoint/register/apple_cryptography.py", line 50, in getAuth
    mobileme = icloud_login_mobileme(
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 40, in icloud_login_mobileme
    g = gsa_authenticate(username, password)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 76, in gsa_authenticate
    r = gsa_authenticated_request(
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 134, in gsa_authenticated_request
    "Request": {"cpd": generate_cpd()},
                       ^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 168, in generate_cpd
    cpd.update(generate_anisette_headers())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 173, in generate_anisette_headers
    h = json.loads(requests.get(config.getAnisetteServer(), timeout=5).text)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/requests/adapters.py", line 519, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='anisette', port=6969): Max retries exceeded with url: / (Caused by NameResolutionError("<urllib3.connection.HTTPConnection object at 0x7d6be11435f0>: Failed to resolve 'anisette' ([Errno -3] Temporary failure in name resolution)"))

saftek1, 2024-12-19:
i am also getting the same error when i try to login to the account for the first time on docker

cyber-vi-king, 2025-02-04:
to me your log reads like it tries to connect to `http://anisette:6969` and fails on the DNS side. Is that correct for your setup? I had a similar problem with anisette running in a separate docker container, so my first try to reach anisette at `localhost` failed similarly.

dchristl, 2025-02-05:
Hello,

please read the documentation and start over. I think @cyber-vi-king's answer is correct. The server can't connect to aniette. Is your anisette server up and running?

zerog2k, 2025-02-08:
I have found that for some reason, `anisette-v3-server` wont run on my old linux ubuntu 20.04 LTS box under docker.
I does not output any logs, and appears to be stuck in a boot loop.

I opened an issue upstream on anisette
https://github.com/Dadoum/anisette-v3-server/issues/32