dchristl/macless-haystack #149: ESP32 Firmware - No Location Updates from Apple Serve

CoderXueXue, 2024-12-29:
Title: ESP32 Firmware - No Location Updates from Apple Server Since 12/27 23:47  

Description:  

On the evening of December 27, I generated keys using `generate_keys.py` and flashed the firmware from the releases page. The device operated normally, and I was able to receive location updates as expected.  

However, on the morning of December 28, I noticed that I stopped receiving location updates from Apple's server. The last update was timestamped at approximately 23:47 on December 27.  

Even as of December 29, I still cannot receive location updates from the server, although the firmware seems to be running without any issues.  

Suspecting a problem with the keys, I regenerated them and reflashed the firmware, but I am still unable to receive updates from Apple's server.  

### Newly Generated Keys  
[image]  

### ESP32 Device Logs  
```
[0;32mI (13492) macless_haystack: Current cycle is 6. Reusing key. [0m
[0;32mI (13492) macless_haystack: Loading key with index 6 at address 169[0m
[0;32mI (13502) macless_haystack: using key with start 68 4d[0m
[0;32mI (13502) macless_haystack: using device address: e8 4d 44 38 3d 0a[0m
[0;32mI (13512) macless_haystack: Sending beacon (with key index 6)[0m
[0;32mI (13532) macless_haystack: advertising has started.[0m
[0;32mI (13622) macless_haystack: stop adv successfully[0m
[0;32mI (13722) macless_haystack: Going to sleep[0m
[0;32mI (13822) macless_haystack: Returned from light sleep[0m
[0;32mI (13822) macless_haystack: Current cycle is 7. Reusing key. [0m
[0;32mI (13822) macless_haystack: Loading key with index 6 at address 169[0m
[0;32mI (13832) macless_haystack: using key with start 68 4d[0m
[0;32mI (13832) macless_haystack: using device address: e8 4d 44 38 3d 0a[0m
[0;32mI (13842) macless_haystack: Sending beacon (with key index 6)[0m
[0;32mI (13862) macless_haystack: advertising has started.[0m
[0;32mI (13952) macless_haystack: stop adv successfully[0m
[0;32mI (14052) macless_haystack: Going to sleep[0m
```

### macless-haystack Logs  
```
2024-12-29 03:19:26,436 - DEBUG - Getting with post: b'{"ids":["2KKAnXNYQcMcrRb+af7OD4SAMeDPkAGJe/KV/9ShMD8="],"days":7}'
2024-12-29 03:19:26,436 - DEBUG - Querying for 7 days
2024-12-29 03:19:27,278 - DEBUG - Return from fetch service:
2024-12-29 03:19:27,278 - DEBUG - {"results":[],"statusCode":"200"}
```  

It seems that the keys are being reused correctly, and the device is sending beacons as expected. However, the server is not returning any results.  

Could this be an issue with the key generation or the beacon configuration? Any guidance on how to troubleshoot this would be appreciated.

J0s3f, 2025-01-09:
Do you have an iPhone or are you sure there are enough iPhones in Bluetooth range?

CoderXueXue, 2025-01-18:
我确定有的，我再重新刷了之后，又恢复了正常，我现在怀疑是否是我只刷入了一个密钥，所以定位只会维持一段时间

> 您有 iPhone 吗？或者您确定蓝牙范围内有足够的 iPhone 吗？

CoderXueXue, 2025-01-18:
> Do you have an iPhone or are you sure there are enough iPhones in Bluetooth range?

我确定有的，我再重新刷了之后，又恢复了正常，我现在怀疑是否是我只刷入了一个密钥，所以定位只会维持一段时间

I'm sure it did, and after I re-swiped it, it's back to normal, and I'm now wondering if I only swiped in one key, so the location will only last for a while

closed 2025-02-14 by CoderXueXue