dchristl/macless-haystack #150: Server offloading

AidanHN, 2024-12-29:
Is there any free solutions to hosting the docker server on a cloud service? I'm assuming if I ran it on my laptop, the laptop would have to stay on for the docker to keep updating the front-end, and I cant be leaving it on 24/7 just to keep getting up-to-date locations. If this is **not** the case, please correct me. Sorry if this sounds stupid, I'm a beginner with servers.
Thanks

CoderXueXue, 2024-12-30:
我是将 dadoum/anisette-v3-server 和 christld/macless-haystack 都部署到了云服务上
我的理解是 这两个服务只是用于从获取位置，即使你不运行服务器，你的固件依旧会正常上报位置
如果docker停止运行，那么和苹果服务器的授权信息将会失效，这样你每次想要查看位置都要重新登陆
我也是服务器新手，但是我很乐意回答你的问题，并且和你一起探讨。

AidanHN, 2024-12-30:
Thanks, that's great to hear 😊. What cloud service did you use? Any information about pricing? 
Thanks

closed 2024-12-30 by AidanHN

AidanHN, 2024-12-30:
Oh ok I understand. Now I need to sort the issue of finding an apple device to connect the new Apple ID to or else it won't be accepted by the Docker. I'll have to ask around..

CoderXueXue, 2024-12-30:
祝你好运

发自我的 iPad