dchristl/macless-haystack #153: Frontend webapllication

satriagilang80, 2025-01-16:
Can you share the source code for a web application? Thanks!

sonictruth, 2025-02-04:
Hi,

If you're interested, I have started a web app client for Macless Haystack. You can check it out here: https://sonictruth.github.io/my-findr/

There is a demo available. Theoretically, you can also connect directly to your API (if there are no bugs) from the Github demo
if you change the settings.
You're welcome to test it.

dchristl, 2025-02-05:
Hello,

what do you mean by source code? You can find the Flutter application [here](https://github.com/dchristl/macless-haystack/tree/main/macless_haystack). The compiled one (html, css, ... ) is inside the webapplication.zip.