dchristl/macless-haystack #156: No updates from device

creuter27, 2025-02-03:
Hi,
I have followed the instructions to the T and everything seemed to work smoothly.
No error messages or problems while flashing the device or installing the server.
Both the server and the ESP32 device (Wroom) seem to work fine as far as I can tell.
I had both my iPad and my Macbook pro lying next to device for hours with bluetooth enabled.
Yet - I never get a location update in my Android client. The request returns with code 200 and no updates every time.

Server interactive mode log:
....
172.20.0.1 - - [03/Feb/2025 18:33:18] "POST / HTTP/1.1" 200 -
2025-02-03 18:48:30,639 - DEBUG - Getting with post: b'{"ids":["5k2v53B6G863O4xYE0YBNUdFUmwKBHveFMQ8nKVzOBA="],"days":7}'
2025-02-03 18:48:30,639 - DEBUG - Querying for 7 days
2025-02-03 18:48:32,407 - DEBUG - Return from fetch service:
2025-02-03 18:48:32,407 - DEBUG - {"results":[],"statusCode":"200"}
172.20.0.1 - - [03/Feb/2025 18:48:32] "POST / HTTP/1.1" 200 -
2025-02-03 19:26:27,093 - DEBUG - Getting with post: b'{"ids":["5k2v53B6G863O4xYE0YBNUdFUmwKBHveFMQ8nKVzOBA="],"days":7}'
2025-02-03 19:26:27,094 - DEBUG - Querying for 7 days
2025-02-03 19:26:27,803 - DEBUG - Return from fetch service:
2025-02-03 19:26:27,803 - DEBUG - {"results":[],"statusCode":"200"}
172.20.0.1 - - [03/Feb/2025 19:26:27] "POST / HTTP/1.1" 200 -
2025-02-03 19:29:39,993 - DEBUG - Getting with post: b'{"ids":["5k2v53B6G863O4xYE0YBNUdFUmwKBHveFMQ8nKVzOBA="],"days":7}'
2025-02-03 19:29:39,994 - DEBUG - Querying for 7 days
2025-02-03 19:29:40,753 - DEBUG - Return from fetch service:
2025-02-03 19:29:40,753 - DEBUG - {"results":[],"statusCode":"200"}
172.20.0.1 - - [03/Feb/2025 19:29:40] "POST / HTTP/1.1" 200 -
...

The app always shows "Fetched 0 locations". 

How can I figure out what the issue is?
Thank you, Chris

creuter27, 2025-02-03:
Update:
I suspected the hardware to be faulty so I pulled out a second ESP32 Wroom device and flashed it wit the same code.
Here is the log:

Serial port COM3
Connecting.........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting..........
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 34:86:5d:fc:73:e0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x000c3fff...
Flash will be erased from 0x00110000 to 0x00110fff...
Compressed 26960 bytes to 16876...
Wrote 26960 bytes (16876 compressed) at 0x00001000 in 2.0 seconds (effective 107.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 114...
Wrote 3072 bytes (114 compressed) at 0x00008000 in 0.1 seconds (effective 204.8 kbit/s)...
Hash of data verified.
Compressed 737168 bytes to 452900...
Wrote 737168 bytes (452900 compressed) at 0x00010000 in 41.4 seconds (effective 142.4 kbit/s)...
Hash of data verified.
Compressed 32 bytes to 41...
Wrote 32 bytes (41 compressed) at 0x00110000 in 0.1 seconds (effective 2.3 kbit/s)...
Hash of data verified.

Leaving...

Seems to have worked just fine. However, again thare aren't any updates.
Does anyone have an idea what I might be doing wrong?

cyber-vi-king, 2025-02-04:
I might have the same problem with two nrf51822 chips. The flashing process looks just about the same as yours. I had one of them connected to gnd and 3.3v for about 2 hours and then flashed the other and left that powered instead and nothing. iPhone sat on the same table about 30-60 cm away. the phone "knew" where it was when I opened maps, but I never received any location for either of the tags.

dchristl, 2025-02-05:
Hello,

could you provide the terminal log of the ESP32? 

Regards,
Danny

lewilou22, 2025-02-06:
My tags have not updated since late january. But my flipper zero is updating

creuter27, 2025-02-06:
False alarm from my side.
The data just needed a very long time to propagate initially.
It is working as expected.

closed 2025-02-06 by dchristl