dchristl/macless-haystack #158: resetting the app following the FAQ and resetting browser data leaves the browser in a state where the flutter app won't load

rabarar, 2025-02-05:
I've followed the instructions on reloading the docker environment, and successfully recreate and authenticate 2FA and restart the mh container. However, when attempting to load the app from the `https://dchristl.github.io/macless-haystack` url, the app only shows the bullseye png and won't load.

I've reset both the browser (chrome) data storage and reset and reverified the docker environment. Is there something else required for the flutter app to run?

dchristl, 2025-02-06:
Hello,

the OperationError indicates that something has been blocked. This could be, for example, plugins that are blocking here. These are usually ad blockers, pop-up blockers, etc. It is best to start a private window and see if it works there. I have tried this with Chrome, Brave and Firefox and the website loaded without any problems.

magnets110, 2025-02-28:
I've had this. if you refresh the page it gets stuck at the logo and give an error. Use developer tools/f12 to clear local storage/cache storage for the site which fixes it