dchristl/macless-haystack #159: pulled new release, reset all, and no longer receive SMS Text for 2FA code

rabarar, 2025-02-06:
running `docker run -it --restart unless-stopped --name macless-haystack -p 6176:6176 --volume mh_data:/app/endpoint/data --network mh-network christld/macless-haystack` prompts for code, but none is ever received and there's no indication of error in the output before the prompt for the 2FA code

dchristl, 2025-02-06:
Does it work before? Do you have more than one phone number in your Apple account? 

> Using phone with id 1 for SMS2FA

This means it is exact the same behaviour like before. This will request the first/only number in your account.

rabarar, 2025-02-06:
It worked consistently before. I only have one phone number associated with the AppleID

dchristl, 2025-02-06:
I'll have a look at it, although I can't explain it at the moment. If you are familiar with Docker and shell, you can try logging in to the Docker container via interactive shell and edit the file /endpoint/register/pypush_gsa_icloud.py. Change the code from line 261 as follows:

```
  # This will send the 2FA code to the user's phone over SMS
    # We don't care about the response, it's just some HTML with a form for entering the code
    # Easier to just use a text prompt
    requests.put(
        "https://gsa.apple.com/auth/verify/phone/",
        json=body,
        headers=headers,
        verify=False,
        timeout=5
    )

    # Prompt for the 2FA code. It's just a string like '123456', no dashes or spaces
    code = input("Enter SMS 2FA code: ")

    body['securityCode'] = {'code': code}
```

Please give feedback afterwards. Thank you 
If you don't trust yourself to do this, I'll try to put the change into a new release soon.

rabarar, 2025-02-06:
Modified the pythons script, deleted and re-ran the container… still no 2FA text

rabarar, 2025-02-07:
so, using `FindMy`  `--trusteddevice` presented for 2FA and generated an `auth.json` file...

dchristl, 2025-02-07:
I have temporarily reset the changes to SM2FA authentication. Restarting the container should be enough.

closed 2025-02-07 by dchristl

magnets110, 2025-02-28:
I cannot get 2FA code either. it worked before. I tried several times and the only way I could get a code was by using FindMy and copying auth.json

dchristl, 2025-03-01:
@magnets110 : The problem is, with the last adjustment the code is identical (apart from a few log outputs) to FindMy. Maybe you should reset everything again.

maxtepafray, 2025-03-03:
@dchristl Your changes didn't fix my issue, but I did find the cause. 

Every time I was re-logging in, I was creating a new device entry on my iCloud account. After about 30 devices Apple's 2FA would start failing and always report a negative during the sms code. Removing the unusaed devices from my account fixed the issue.

dchristl, 2025-03-09:
@maxtepafray 
Thank you, I will add it to the Readme