dchristl/macless-haystack #160: Hello, is ESP32-C6 supported?

windyslime, 2025-02-07:
I tried to run this on ESP32-C6 and it can't run. I wonder if there is a problem about support. If ESP32-C6is unsupported which model should I buy?

dchristl, 2025-02-08:
Hello, 

the ESP32 C6 is not supported. You can use a "normal" ESP32, without Sxx or Cxx.

windyslime, 2025-02-09:
Thanks for reply. But is there a way to use ESP32-C6 in this project? Like compiling from source code or something else. I'm in a hurry. This product is for a contest that closes on February 17th. And I don't I can brought a new one in time.

dchristl, 2025-02-09:
Try this: [esp32-c6.zip](https://github.com/user-attachments/files/18723933/esp32-c6.zip)

Unfortunately, I can't offer you any support if something doesn't work