dchristl/macless-haystack #163: Location decryption seems to have a Problem.

Systm21, 2025-02-09:
[image]

I updated everything today, even though the old server was running perfectly, but now I have a very strange problem. The location of the first tracker that received a report after the update is at the end of the world and completely out of order. 

At first I thought it was due to the receiver device, that there was a problem with the GPS reception, but I get correct data when I use the request_reports.py from biemster. I can't explain this either, the same problem occurs both in the Android app and on https://dchristl.github.io/macless-haystack/. 

The old entries also don't have the battery next to the timestamp, could this be related to this? Has this change caused a problem?

The location point should actually be in Europe.

dchristl, 2025-02-11:
Unfortunately, I also occasionally have such points (once a month) and they are actually reported by Apple. The code is already filtered out based on confidence. This normally takes care of itself when a new report is reported.  
What surprises me is that you get the correct coordinates via request_reports. The data is basically the same. 

I would need the raw data (preferably the JSON file) to be able to analyse this in more detail

supaeasy, 2025-02-11:
It happened to me today as well for the first time for one device. I did not touch the tracker's firmware; I flashed it about a year ago. It also shows a low battery sign, but I think that feature wasn't even implemented back then, right?

Systm21, 2025-02-12:
As already mentioned, the problem only seems to be temporary. It has never occurred in the past, but is not very bad now.

josemariaaraujo, 2025-02-28:
I'm having the same problem. "request_reports.py" give me the correct data, macless-haystack sends the device to the arctic:
[image]

```
root@testes:~/FindMy# ./request_reports.py -p CMD -H 27  
pyprovision is not installed, querying http://192.168.1.2:6969 for an anisette server
200: 100 reports received.
7 reports used.
{'lat': 41.161041, 'lon': -8.6497215, 'conf': 166, 'status': 0, 'timestamp': 1740675886, 'isodatetime': '2025-02-27T17:04:46', 'key': '_G8dt9BR', 'goog': 'https://maps.google.com/maps?q=41.161041,-8.6497215'}
{'lat': 41.1662961, 'lon': -8.6429986, 'conf': 118, 'status': 0, 'timestamp': 1740675902, 'isodatetime': '2025-02-27T17:05:02', 'key': '_G8dt9BR', 'goog': 'https://maps.google.com/maps?q=41.1662961,-8.6429986'}
{'lat': 41.1806106, 'lon': -8.6572446, 'conf': 129, 'status': 0, 'timestamp': 1740676083, 'isodatetime': '2025-02-27T17:08:03', 'key': '_G8dt9BR', 'goog': 'https://maps.google.com/maps?q=41.1806106,-8.6572446'}
{'lat': 41.1835032, 'lon': -8.6621267, 'conf': 140, 'status': 0, 'timestamp': 1740676162, 'isodatetime': '2025-02-27T17:09:22', 'key': '_G8dt9BR', 'goog': 'https://maps.google.com/maps?q=41.1835032,-8.6621267'}
{'lat': 41.2183454, 'lon': -8.6936288, 'conf': 104, 'status': 0, 'timestamp': 1740676590, 'isodatetime': '2025-02-27T17:16:30', 'key': '_G8dt9BR', 'goog': 'https://maps.google.com/maps?q=41.2183454,-8.6936288'}
{'lat': 41.2182364, 'lon': -8.6934339, 'conf': 97, 'status': 0, 'timestamp': 1740676665, 'isodatetime': '2025-02-27T17:17:45', 'key': '_G8dt9BR', 'goog': 'https://maps.google.com/maps?q=41.2182364,-8.6934339'}
{'lat': 41.2156894, 'lon': -8.691441, 'conf': 211, 'status': 0, 'timestamp': 1740676903, 'isodatetime': '2025-02-27T17:21:43', 'key': '_G8dt9BR', 'goog': 'https://maps.google.com/maps?q=41.2156894,-8.691441'}
found:   ['_G8dt9BR']
missing: []
root@testes:~/FindMy#
```
I was driving at the time with the device. The last timestamp is the same, the coordinates are not.

Find attached the JSON of my device. I've cleaned the cookies and re-added it to macless and was able to reproduce the problem.
[accessories (1).json](https://github.com/user-attachments/files/19032650/accessories.1.json)

dchristl, 2025-03-01:
I think I have found the error using your JSON. Thank you @josemariaaraujo (I have removed the file for privacy reasons). It seems like certain reports are encoded differently, which is why the decoding determines different coordinates. I have provided a fix for this and if you are using the web version (GitHub) then this should work. You may need to reset the Accessory again. The Android version will be fixed in the next release.

Systm21, 2025-03-01:
I had the problem again in my timeline, your fix works great in the web version, look forward to when the android version is fixed too. Many thanks for that :)

dchristl, 2025-03-01:
@Systm21 : If you can't wait (I don't know, when I have time for a release), you can use the [build artefacts ](https://github.com/dchristl/macless-haystack/actions/runs/13608114739) at the build pipeline. (you must be logged in GitHub)

Systm21, 2025-03-01:
I have to revise my statement again. Normally it should no longer generate the old points, should it? But it definitely still does. Both in the web and in the android app...

[image]

dchristl, 2025-03-01:
Have you resetted your accessory (there is a new option in accessory managemement).?  This can be an old history entry.  Alternatively you can remove the accessory and add it again.

Systm21, 2025-03-01:
Reset button works perfectly and the fix does exactly what it should. 
Could you perhaps add a “Reset all accessories” button in the settings?

dchristl, 2025-03-01:
> Reset button works perfectly and the fix does exactly what it should.
Could you perhaps add a “Reset all accessories” button in the settings?

Not planned, yet. because this will be solved automatically after 7 days ;)

dchristl, 2025-03-03:
A new release is available, solving this issue.

closed 2025-03-03 by dchristl