dchristl/macless-haystack #164: Tags imported via JSON are not saved in APP v2.3.0

Systm21, 2025-02-12:
I updated from 2.2.0 to 2.3.0, which was no problem so far, everything worked as it should. When I wanted to import new tags via json, I noticed that they were no longer available after restarting the app. This had never happened before in 2.2.0...
To reproduce the problem, I deleted all the tags from the app and imported them all again. After a restart, none of the tags remained, they were all gone and stayed gone. 
I then deleted the app and installed 2.2.0, which is no longer compatible, but here I was able to import my tag collection plus the new tags. I then updated to 2.3.0 again. Funnily enough, they all remained saved. (But I still can't import new ones)

Systm21, 2025-02-12:
Problem is NOT present in the current version of the web app!

Systm21, 2025-02-13:
I have just realized that the problem also exists in relation to the reports. When opening the app, all tags are in status unknown and must first be retrieved, previously this was also saved.

dchristl, 2025-02-13:
This is a duplicate of #161 and is caused by an issue in an external library. I will release a new version soon. In the meantime you can use the apk in #161

closed 2025-02-13 by dchristl