dchristl/macless-haystack #169: Devices getting mixed up

potchin, 2025-02-18:
I bought 2 nRF51822 devices, followed the instructions to generate keys, patch firmware files and then flashed them. I have 2 separate folders which contain different keys, firmwares etc. I'm 99% sure I flashed different files to each device.. but when I track the devices I only get location data for 1 device (which can actually be either physical device!).

Firstly I want to make sure I didnt do something silly like flash the same firmware to multiple devices - is there a way to figure out which firmware has been flashed via BLE? The solder pads are tiny on my devices and resoldering them back up is likely to cause damage. Surely they are broadcasting something which I can use to figure out the key being used? I dont see either in "nRF Connect" on Android.

Assuming I didnt flash the same firmware twice, is there anything else in the setup which could cause this issue? I'm running my own annisette server alongside v2.5.0 of macless-haystack (just upgraded from the previous version)

dchristl, 2025-02-18:
You could, for example, compare the firmwares using a hash (md5) if you still have the files. If they are identical, then you have used the same one twice. I suspect you did this because you probably didn't re-download the raw firmware in between. You must always apply the xxd command to the original firmware, otherwise the key won't be replaced. You can recognize this by the output. If it says

 ```
0+0 records in
0+0 records out
```

the second time, then nothing is being replaced and you will have an unchanged firmware.

potchin, 2025-02-18:
Yep, I already checked the md5 of each file and they are different. I just
need a way to check that I didn't accidentally flash the same file to both
devices.

Should nRF connect show them? Not seeing either on my pixel 8

dchristl, 2025-02-18:
Normally nrf connect should show them, but  this doesn't work for me either. Some users reported success with it. You can try Ble smart scanner. Maybe you can see differences, but I think you have flashed the firmware twice. I have multiple devices and this never happened and no user had this issue before. You can recognize an error here very quickly.

potchin, 2025-02-18:
Yep, I did something silly and flashed the same firmware to two devices somehow. Reflashed a new firmware and it seems to be working. Thanks 🤦

closed 2025-02-18 by potchin