dchristl/macless-haystack #171: Why flutter app fetch locations slower than `request_reports.py`

lovelyelfpop, 2025-02-20:
200 keys, it takes 7 seconds to fetch location reports for the app(from app fully launched to "Fetched xx location(s)" toast), but it only takes 2 seconds for `request_reports.py`(include fetch reports and decrypt them).

Is it because flutter is slower than python? 

It seems that it takes 3 seconds for the app to generate public keys from private keys in `PREFIX_devices.json`. What if use `PREFIX.keys` instead of `PREFIX_devices.json`, the time to generate the public keys can be saved.

dchristl, 2025-02-23:
Flutter is not generally slower than python, but usually even faster. Flutter generates and compiles code for the respective target system (Android -> Java, Linx -> C++, Web ->Javascript, ...). Javascript (web) is much slower compared to python all others are much faster. However, additional optimisations have already been added, e.g. reports that have already been decoded are not decoded again. It is therefore normally only very slow the first time, but fast afterwards. 
But why do you use 200 keys at all, for tracking reasons (rotating keys ) 10 or 20 are completely sufficient. This also generally slows the whole thing down a lot, because all 200 keys have to be queried.  

> It seems that it takes 3 seconds for the app to generate public keys based on private keys in PREFIX_devices.json. What if use PREFIX.keys instead of PREFIX_devices.json, the time to generate the public keys can be saved.

That can't be right or I don't understand what you mean. The main problem from a performance point of view is the decoding of the reports and the web service query. Which in turn leads me back to the fact that you should simply use far fewer keys

lovelyelfpop, 2025-02-24:
I have plenty of tags, so there are 200 keys. I debuged the code of flutter app on android, It did took 3 seconds to `generateHashedPublicKeys`(`lib\accessory\accessory_registry.dart`), and 4 seconds to `await Future.wait(runningLocationRequests)`

dchristl, 2025-03-01:
Sorry, but I don't understand what you mean and what you have debugged. There is no generateHashedPublicKeys method in the file. In my mass tests (Android emulator) I have about 9500 reports with 20 keys. With a new (not yet cached accessory), this initially takes approx. 1 minute, with decryption taking approx. 10 - 20 ms per report. The rest is negligible. And this is only the first time, with each subsequent call, reports that have already been decrypted are ignored. As a user, however, you don't really notice this because the older reports are only used to build up the history.

lovelyelfpop, 2025-03-03:
Sorry, my bad.
What I mean is the code below in`lib\accessory\accessory_registry.dart` takes 3 seconds to generate publilc keys when there are 200 keys(a lot of tags)
```
for (var i = 0; i < currentAccessories.length; i++) {
      var accessory = currentAccessories.elementAt(i);

      var keyPair =
          await FindMyController.getKeyPair(accessory.hashedPublicKey);

      List<FindMyKeyPair> hashedPublicKeys =
          await Stream.fromIterable(accessory.additionalKeys)
              .asyncMap((hashedPublicKey) =>
                  FindMyController.getKeyPair(hashedPublicKey))
              .toList();

      hashedPublicKeys.add(keyPair);

      var locationRequest =
          FindMyController.computeResults(hashedPublicKeys, url);
      runningLocationRequests.add(locationRequest);
    }
```

dchristl, 2025-03-03:
Ok, I understand your problem. In my tests I never really noticed this, as the complete key generation takes 9ms on average. That is something (and with 200 keys about 1.8s) but not the main part. I'll see if I can speed it up a bit by storing the required keys, but that means that the amount of data per key triples (2 additional keys have to be stored) and reading them out of the SecureStorage also costs something. 

>  It takes too much time to fetch location reports of all accessories at once, and to parse and display the geo location only after obtaining the reports of all accessories.
Also, if there are a lot of accessories, it would be nice if I could search and locate a specific accessory.

I could imagine making it possible to deactivate fetching during startup via a setting and at the same time realise an update of a single tag (e.g. via longTap). That would solve at least some of your problems.

dchristl, 2025-03-03:
I've implemented the possibility to update one accessory. Check out the latest release.