dchristl/macless-haystack #172: Error getting location

maximushugus, 2025-02-23:
I managed to connect with 2FA my macless-haystack docker.

I then created a new accessory and flash an ESP32 board with Advertisement key.
On the website I put my macless-haystack server IP in configuration.

But when i try to refresh the website to get device location I  have this error : 
`Could not find location reports. Try again later. Error: ClientException: NetworkError when attempting to fetch resource., uri=http://myip:6176`

In macless-haystack logs I have : 
```shell
2025-02-23 17:01:33,272 - DEBUG - Getting with post: b'{"ids":["gIFxdxxxxxxxxx4gl3Au1zCO0ojHm0fO5/hcg9Urk="],"days":7}'
2025-02-23 17:01:33,273 - DEBUG - Querying for 7 days
2025-02-23 17:01:34,324 - DEBUG - Return from fetch service:
2025-02-23 17:01:34,325 - DEBUG - 
2025-02-23 17:01:34,325 - ERROR - Unknown error occurred Expecting value: line 1 column 1 (char 0)
Traceback (most recent call last):
  File "/app/endpoint/mh_endpoint.py", line 103, in do_POST
    result = json.loads(r.content.decode())
  File "/usr/local/lib/python3.13/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ~~~~~~~~~~~~~~~~~~~~~~~^^^
  File "/usr/local/lib/python3.13/json/decoder.py", line 345, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/json/decoder.py", line 363, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
192.168.3.131 - - [23/Feb/2025 17:01:34] "POST / HTTP/1.1" 501 -
``` 

Do you know why I get this error ?
thanks !

dchristl, 2025-02-23:
Hello, 

it seems your container is getting an invalid response (maybe empty) from Apple. You can try to set the loglevel to DEBUG in the config.ini  to see the exact response: 
`loglevel=DEBUG`

You can also check if  your container has access to https://gateway.icloud.com/acsnservice/fetch ?

maximushugus, 2025-02-23:
I'm already on DEBUG log level.
Doing a curl https://gateway.icloud.com/acsnservice/fetch inside the docker container doesn't return error so I think it's ok

dchristl, 2025-02-23:
Sorry, I see you are already on debug. After `Return from fetch service:` there is normally the output if there is any.  It is hard to tell why there is an empty response. I would just [reset](https://github.com/dchristl/macless-haystack/blob/main/FAQ.md#how-can-i-reset-everything-and-start-over-how-can-i-completely-uninstall-macless-haystack) and start over.

maximushugus, 2025-02-23:
I also tried reseting and connecting again.
I forgot to say that I connected using my VPS and copied auth.json to my home server. I have no error lauching macless-haystack inside docker in my home server, so I think the connection to apple is ok.

maximushugus, 2025-02-23:
Would it be possible that if my device has never been seen apple (low density), appel returns nothing and this give me this error ?

dchristl, 2025-03-01:
> I also tried reseting and connecting again. I forgot to say that I connected using my VPS and copied auth.json to my home server. I have no error lauching macless-haystack inside docker in my home server, so I think the connection to apple is ok.

You may want to rule out sources of error. Does macless haystack run with your account if you do not start it on the VPS, but locally, for example?

maximushugus, 2025-03-27:
I managed to solve the issue by reconnecting my container to apple id.
On unraid, to be able to launch the container interactively, you have to stop the docker service, and on docker settings enable Template Authoring Mode.
Then create your custom docker template and save it. You will see the command unraid uses to create this container starting with /usr/local/emhttp/plugins/dynamix.docker.manager/scripts/docker create --name='macless-haystack' ...

Copy this command and add --interactive --tty at the end of the command just before the repository ('christld/macless-haystack').

Execute the modified command under unraid cli.

Then do "docker start -ai <container-id>" on cli and here you can enter your apple ID, and 2FA.
Then close this cli and relauch the container via webgui.

closed 2025-03-27 by maximushugus