dchristl/macless-haystack #174: After first pickup location next pickup is after at least few minutes

tomasz90, 2025-03-03:
Hi!

Original airtag sends location every 2s. After setting up our tag i noticed that even if tag is advertising with the same frequency, location is only send every few minutes 10-20 minutes. Is that intentional? I use esp32 with the android app.

tomasz90, 2025-03-04:
But in 20 minutes there is 20 of minutes ^^

I also modified ESP32 code to sends every 2s and it still doesnt make any difference, location is picked up in a very long intervals.

dchristl, 2025-03-09:
The reception of the locations depends not only on how often the device transmits, but also on whether someone with an iPhone picks up and forwards the signal at that exact moment. If you take your device into a busy city centre, the number of locations should increase. Alternatively, you can also adjust the code so that it corresponds to the [original](https://github.com/seemoo-lab/openhaystack/blob/main/Firmware/ESP32/main/openhaystack_main.c). This transmits continuously, which increases the number of locations but is almost impossible to operate with a battery.