dchristl/macless-haystack #177: Macless Haystack Login Issue

Minerva4545, 2025-03-07:
I've been experiencing issues running Macless Haystack for the past few days. Previously, after entering my Apple ID and password, I would receive a verification code and could start using it immediately. However, even though I still receive the code, I'm now having trouble running it.

Here is the error message I get:

`Enter SMS 2FA code (If you do not receive a code, wait 60s and press Enter. An attempt will be made to request the SMS in another way.):
Enter SMS 2FA code:504595
Traceback (most recent call last):
  File "/app/endpoint/mh_endpoint.py", line 177, in <module>
    apple_cryptography.registerDevice()
  File "/app/endpoint/register/apple_cryptography.py", line 71, in registerDevice
    getAuth(regenerate=True)
  File "/app/endpoint/register/apple_cryptography.py", line 44, in getAuth
    mobileme = icloud_login_mobileme(
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 45, in icloud_login_mobileme
    g = gsa_authenticate(username, password)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 126, in gsa_authenticate
    sms_second_factor(spd["adsid"], spd["GsIdmsToken"])
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 301, in sms_second_factor
    resp.raise_for_status()
  File "/usr/local/lib/python3.12/site-packages/requests/models.py", line 1021, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 401 Client Error:  for url: https://gsa.apple.com/auth/verify/phone/securitycod
`

I have also tried using older versions, but the issue persists.

Could you please help me resolve this?

Thank you!

maxtepafray, 2025-03-07:
@Minerva4545 Open up your icloud and check your devices. I ran into a similar issue where I couldn't 2fa after logging in too many times. Seems like the current implementation uses a new device id each time, and it fills up your account with devices.

After so many, 2fa just refuses to work. Go in and delete all the devices you're not using. They should appear as 13in Macbooks

dchristl, 2025-03-09:
Have you ever tried the script from [biemster](https://github.com/biemster/FindMy)? If the login works here, then you can also use the resulting auth.json for macless-haystack

isibizi, 2025-04-29:
I have tried it with biemsters script and i dont receive any 2fa codes