dchristl/macless-haystack #187: is there any way to disable the light that shows up on the microbit?

aaronjrodrigues, 2025-05-01:
I wanted to optimise the microbit's battery by getting rid of the power light that shows up, thanks

nekename, 2025-05-06:
Hello @aaronjrodrigues, I just wanted to ask how you actually got this working. I flashed the .bin file like you usually do to a microbit but I don't get a light like you do.

aaronjrodrigues, 2025-05-06:
Just want to confirm, is your open haystack flashed and working? And along with that your power light isn't on? I flashed mine by just placing the .bin file inside the microbit. Is that how you did it?

aaronjrodrigues, 2025-05-06:
copy @nekename

nekename, 2025-05-07:
Hey @aaronjrodrigues , thanks for replying. To be honest, I think that my issue was just that there weren't enough iPhones around, because it seems to be working now.

As for the power LED, I'm powering it with a battery pack that uses the 3V and GND pins - so maybe the LED comes on only when powered from the battery header?

aaronjrodrigues, 2025-05-07:
> Hey [@aaronjrodrigues](https://github.com/aaronjrodrigues) , thanks for replying. To be honest, I think that my issue was just that there weren't enough iPhones around, because it seems to be [...]

do you mind sending me a picture as to how its setup and/or a guide on how to get this done? would love to set it up like that if it means not having the light on. thanks

nekename, 2025-05-08:
[image]

I'm using one of these I had lying around from a very long time ago, it's a microbit v1.5