dchristl/macless-haystack #190: Issue with not receiving 2FA code for Chinese users

Lonelymo, 2025-05-15:
I am a user from China. I tried to run the server using Docker. After entering my Apple ID and password, I reached the 2FA verification step but didn't receive the code.

**Possible Cause:**
I discovered that iCloud in China differs from iCloud abroad—one uses the .com domain and the other uses .com.cn. To address this, I used an Apple device abroad and an international Apple ID and password, but still didn't receive the 2FA code.

**Error Message:**
After waiting for 60 seconds and pressing Enter, the following error appeared:

[image]

**Steps to Reproduce:**
1. Run the server using Docker.
2. Enter Apple ID and password.
3. Reach the 2FA step but don’t receive the code.
4. Use an international Apple device and Apple ID, but still don’t receive the 2FA code.

Lonelymo, 2025-05-15:
[image]

Yxini-p, 2025-05-23:
我是用国行的账号的；添加了受信任电话后，你退出苹果登录的账号；就会发短信到你的信任号码的；gitee有个类似的项目你可以看一下；

num13ru, 2025-06-20:
Same for the Apple ID with RU region

primec0de, 2025-08-17:
Same error for Apple ID with US region:
```
   raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 401 Client Error:  for url: https://gsa.apple.com/auth/verify/phone/securitycode
```

Shxuai, 2025-09-16:
Same error for Apple ID with CN region:
`2025-09-16 08:38:40,500 - INFO - Authentication request initialization
Traceback (most recent call last):
  File "/app/endpoint/mh_endpoint.py", line 177, in <module>
    apple_cryptography.registerDevice()
  File "/app/endpoint/register/apple_cryptography.py", line 71, in registerDevice
    getAuth(regenerate=True)
  File "/app/endpoint/register/apple_cryptography.py", line 44, in getAuth
    mobileme = icloud_login_mobileme(
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 45, in icloud_login_mobileme
    g = gsa_authenticate(username, password)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 82, in gsa_authenticate
    r = gsa_authenticated_request(
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 157, in gsa_authenticated_request
    resp.raise_for_status()
  File "/usr/local/lib/python3.12/site-packages/requests/models.py", line 1021, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 503 Server Error: Service Temporarily Unavailable for url: https://gsa.apple.com/grandslam/GsService2
`

ch155260, 2025-09-16:
I got same problem, does any one have method to solve the problem?

> 2025-09-16 08:40:49,448 - INFO - Using phone with id 1 for SMS2FA
>Enter SMS 2FA code (If you do not receive a code, wait 60s and press Enter. An attempt will be made to request the SMS in another >way.):
>2025-09-16 08:40:52,258 - INFO - You only waited 2 seconds. The next request will be started in 58 seconds
>Enter SMS 2FA code if you have received it in the meantime, otherwise press Enter: Traceback (most recent call last):
>   File "/app/endpoint/mh_endpoint.py", line 177, in <module>
>     apple_cryptography.registerDevice()
>     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
>  File "/app/endpoint/register/apple_cryptography.py", line 71, in registerDevice
>    getAuth(regenerate=True)
>    ~~~~~~~^^^^^^^^^^^^^^^^^
>  File "/app/endpoint/register/apple_cryptography.py", line 44, in getAuth
>    mobileme = icloud_login_mobileme(
>        username=mh_config.getUser(), password=mh_config.getPass())
>  File "/app/endpoint/register/pypush_gsa_icloud.py", line 45, in icloud_login_mobileme
>    g = gsa_authenticate(username, password)
>  File "/app/endpoint/register/pypush_gsa_icloud.py", line 126, in gsa_authenticate
>    sms_second_factor(spd["adsid"], spd["GsIdmsToken"])
>    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>  File "/app/endpoint/register/pypush_gsa_icloud.py", line 287, in sms_second_factor
>    code = request_code(headers)
>  File "/app/endpoint/register/pypush_gsa_icloud.py", line 329, in request_code
>    req.raise_for_status()
>    ~~~~~~~~~~~~~~~~~~~~^^
>  File "/usr/local/lib/python3.13/site-packages/requests/models.py", line 1026, in raise_for_status
>    raise HTTPError(http_error_msg, response=self)
>requests.exceptions.HTTPError: 405 Client Error:  for url: https://gsa.apple.com/auth/verify/phone/

Shxuai, 2025-09-17:
> I am a user from China. I tried to run the server using Docker. After entering my Apple ID and password, I reached the 2FA verification step but didn't receive the code.
>
> **Possible Cause:** I [...]

感觉不是iCloud账号区域原因，我用美区的icloud账号也不能登入。我怀疑是因为我用的阿里云服务器部署的，可能苹果的风控把阿里的请求都ban了。不知道你用什么方式部署的？

Tokisaki-Galaxy, 2025-09-27:
> > 我是来自中国的用户。我尝试使用 Docker 运行服务器。输入我的 Apple ID 和密码后，我到达了 2FA 验证步骤，但没有收到验证码。
> > **可能的原因：**我发现中国的 iCloud 与国外的 iCloud 不同——一个使用 .com 域名，另一个使用 .com.cn。为了解决这个问题，我在国外使用了 Apple 设备以及国际 Apple ID 和密码，但仍然没有收到 2FA 代码。
> > **错误信息：**等待 60 秒并按 Enter 后，出现以下错误：
> > [image]
> > **复制步骤：**
> > 
> > 1. 使用 Docker 运行服务器。
> > 2. 输入 Apple ID 和密码。
> > 3. 到达 2FA 步骤，但未收到代码。
> > 4. 使用国际 Apple 设备和 Apple ID，但仍未收到 2FA 代码。
> 
> 感觉不是iCloud账号区域原因，我用美区的icloud账号也不能登入。我怀疑是因为我用的阿里云服务器部署的，可能苹果的风控把阿里的请求都ban了。不知道你用什么方式部署的？

阿里云和腾讯云的服务器IP段似乎都被ban了，但是被ban的提示是503服务不可用，`curl -k https://gsa.apple.com/grandslam/GsService2 -v`可以通过这个测试是否被ban