dchristl/macless-haystack #194: Apple seems changed something this week, multiple keys request is limited to 20~40 reports and it all gets filled by one of the keys

lovelyelfpop, 2025-05-25:
see https://github.com/biemster/FindMy/issues/82

No wonder I can’t get the latest location for almost all my tags.

lovelyelfpop, 2025-05-28:
I modified request_reports.py. I use async and parallel requests to save time. See:
https://gist.github.com/lovelyelfpop/9733899e3793df3b74d1da57095c46ce

zen85, 2025-06-26:
> request_reports.py

i cant find the original request_reports.py in the repository? are you sure you are in the right git?

lovelyelfpop, 2025-06-27:
My bad, `request_reports.py` is in https://github.com/Chapoly1305/FindMy/ and https://github.com/biemster/FindMy/

New `mh_endpoint.py` is here 
https://gist.github.com/lovelyelfpop/a6c31aacdfb309103b2938687aec04ce

josephBoonrawd, 2025-08-25:
@lovelyelfpop  I run
docker start -ai macless-haystack
From https://github.com/dchristl/macless-haystack
 * branch            main       -> FETCH_HEAD
Traceback (most recent call last):
  File "/app/endpoint/mh_endpoint.py", line 14, in <module>
    from aiohttp import web, BasicAuth, ClientSession
ModuleNotFoundError: No module named 'aiohttp'

have error look like this when I change mh_endpoint.py of macless is here
https://gist.github.com/lovelyelfpop/a6c31aacdfb309103b2938687aec04ce

lovelyelfpop, 2025-08-26:
@josephBoonrawd `aiohttp` module is required. `pip3 install aiohttp` or edit `requirements.txt`

lovelyelfpop, 2025-08-27:
@Stardawn0v0 If a tag flashed firmware with a single key  is stick together with the same iPhone, the iPhone will only report its location at the beginning and will almost never upload its locations afterwards. I guess this is related to key rotation. The location of my other tags with multiple keys get updated more frequently.  AFAIK, only nrf5x's firmware support multiple keys. I inject 228 keys for each one of my tags, and KEY_ROTATION_INTERVAL=3000ms, so it can be rotated for seven days without duplication.

lovelyelfpop, 2025-09-01:
@josephBoonrawd I create a new docker image "lovelyelfpop/macless-haystack-gitee". Here is the new command:
```
docker run -it --restart unless-stopped --name macless-haystack-gitee -p 6176:6176 --volume mh_data:/app/endpoint/data --network mh-network lovelyelfpop/macless-haystack-gitee
```

josephBoonrawd, 2025-09-15:
@lovelyelfpop if I don't reload Host: Browse location no update, I must reload Host: Browse every 2 min for update location

How to  fix?

lovelyelfpop, 2025-09-16:
@josephBoonrawd  Press the refresh button at the bottom-right

josephBoonrawd, 2025-09-17:
@lovelyelfpop If I do not refresh, the location will not be saved in the history. Is that correct?

I must code for refresh every 2 min for saved in the history.

and sometime I have error location look like this

[image]
Have you ever encountered it?

lovelyelfpop, 2025-09-17:
@josephBoonrawd Yes, I have, but not that far, 1km at most. I don't know why, maybe the confidence of this location report is low

josephBoonrawd, 2025-09-17:
@lovelyelfpop My error more than 1 km.

I have issue

If I do not refresh, the location will not be saved in the history. Is that correct?

I must code for refresh every 2 min for saved in the history.

lovelyelfpop, 2025-09-18:
@josephBoonrawd 
```
- "If I do not refresh, the location will not be saved in the history. Is that correct?"
- Yes, the app needs to actively request the Apple server to obtain location reports. But each refresh will get the location reports of the last few days, no need to refresh every 2 minutes 
```

lovelyelfpop, 2025-09-18:
@josephBoonrawd I am using pix's firmware too. It's a problem of macless-haystack app. Comment this line https://github.com/dchristl/macless-haystack/blob/main/macless_haystack/lib/findMy/decrypt_reports.dart#L82 and build web/apk for your own.

josephBoonrawd, 2025-09-19:
@lovelyelfpop 

[image]

[image]

How to build web/apk for my own.