dchristl/macless-haystack #210: Can't setup web application

daniwyder, 2025-07-22:
Hi, first of all I want to apologize in advance for this rookie question but I'm very new to web applications. I have a little problem setting up the web application. The server is running on another machine which I'm connected to via ssh and I'm trying to get the web application going on my machine. Unfortunately I don't even have an idea on how to configure it so it will work and I'm not able to set it up based on the description in the readme. Can you please help me a little or tell me which files I have to change for the configuration? Thank you very much in advance.

ReLoia, 2025-07-25:
hi, I think the discussions page is better suited for a question like this.

Anyway,
did you execute the commands in the readme?
```bash
docker network create mh-network

docker run -d --restart always --name anisette -p 6969:6969 --volume anisette-v3_data:/home/Alcoholic/.config/anisette-v3 --network mh-network dadoum/anisette-v3-server

docker run -it --restart unless-stopped --name macless-haystack -p 6176:6176 --volume mh_data:/app/endpoint/data --network mh-network christld/macless-haystack
```

When executing the second command it will prompt you for your apple id (your email) and apple password and then it will wait for you to submit the MFA code from apple.

Once done that it will close the execution of the docker instance and you will simply need to restart macless-haystack's frontend.

```bash
docker restart macless-haystack
```

Maybe the problem is related to accessing the webpage remotely? Do you have experience with Reverse Proxies like Nginx or Caddy or with port opening?

daniwyder, 2025-07-30:
Sorry, should have thought about that regarding the discussion page. Yes I did everything exactly like in the description but I'm sure the problem is on the remote access. I managed to just make it run locally and that's fine for now. Thank you anyway for your help @ReLoia :)

closed 2025-07-30 by daniwyder